bigtreetech/BIGTREETECH-Stepper-Motor-Driver #8: S42B v2 firmware not working

rocko1337, 2021-05-04:
The firmware of S42B v2 can be found here:
BIGTREETECH-Stepper-Motor-Driver/S42B/firmware/

I flashed it using stlink but it is not working (motor is not spinning correctly). I also compiled it from source code, but I have the same problem. The source code is not working properly.

Bigtreetech, can you please upload the working source code? I would like to adjust the code to send the current position (angle) over uart. Thank you.

jubjub64, 2021-05-04:
I am also having this same issue with compiling. The compiled bin (using the code found here: [LINK](https://github.com/bigtreetech/BIGTREETECH-Stepper-Motor-Driver/tree/master/S42B/firmware/S42BV2.0) ) is much smaller than the bin that was originally on the S42B v2.

rocko1337, 2021-05-04:
After flashing the firmware the motor is bricked. Luckily I have two motors and I dumped the firmware from the functioning one (STLINK) and transfered it to the bricked motor, after that the bricked motor worked again. I am 100% sure the source code on github is old and was not updated. Please update the code bigtreetech. Thank you.

jubjub64, 2021-05-19:
I got this from messaging "Btt Tony" on their Facebook page: "Hi, the S42B firmware on our official website is currently an old version, there is a problem. I have given feedback to our engineers and are planning an update."

svsergo, 2021-07-08:
When will the corrected firmware be released?  To adapt to the engine with a step of 0.9 degrees, the original working code is required.

rocko1337, 2021-07-10:
> I got this from messaging "Btt Tony" on their Facebook page: "Hi, the S42B firmware on our official website is currently an old version, there is a problem. I have given feedback to our engineers and are planning an update."

Thank you jubjub64 for asking on facebook! Can you please poke Btt Tony again? I really would like to get the working source code.

jubjub64, 2021-07-10:
> Thank you jubjub64 for asking on facebook! Can you please poke Btt Tony again? I really would like to get the working source code.

I just sent him a request for another update on when it will be fixed. Will report back if/when I get an answer.

jubjub64, 2021-07-12:
I heard back from Btt Tony:
"Hi, I'm sorry, I asked him again, he said that he is busy developing S42T recently. He will update the firmware as soon as possible."

rocko1337, 2021-10-07:
Would be really nice if Btt Tony could finally upload it :-|

rob-mcinerney, 2022-01-16:
I am having the same problem, did this ever get resolved?

rocko1337, 2022-02-08:
> I am having the same problem, did this ever get resolved?

Unfortunately not. We are still waiting :(

rocko1337, 2022-12-26:
Still waiting.

cklopotic, 2023-06-21:
Hello?  Has anyone got the updated source code for this yet?