bigtreetech/BIGTREETECH-Stepper-Motor-Driver #11: s42b v1.1 - stepping inacuracy on inner edges

linewallker, 2021-11-30:
Hello,
  I am using S42b v1.1 servo for my CNC machine, problems with PID settings I have solved, but I am experiancing very strange things.

When I am drawing rectangle with black infill it is ok, outer edges are smooth and precise for every row, can perform cutting speed more than 1500 mm/min with no problems.

But when I am trying to create some TEXT inside the rectangle, the edges of this text have side deviation about 1 MM for each side( dependatnt if it is from right to left or left to right, every row have oposite deviation than the row before, it is making 2 mm blured edge on the engraved picture, issue is starting to be problem in smaller plots, in bigger pictures it is not so visible, but the problem is still there.

Can you please tell me if I can update servo S42B v1.1 wizh 2.0 firmware and also can you please point me where I should look ?

I am starting to be out of ideas.

Changed software from GRBL to GRBL ESP32 than FLUIDNC with no change on the problem, also I can change what I want in the engraver settings with no affect on deviation, I can change speeds, acceleration, junction deviation, pulse delay and so on with absolutely no change on the problem.

Please help, I need to get the machine operational asap, xmass are almost here and I have a lot orders for products, but I can not engrave small things because of this problem.

I will post photos after I will get back from Office.

linewallker, 2021-11-30:
[image]

linewallker, 2021-12-01:
Hello,
  Can somebody please answer my question?
Really do not know what I should look for

linewallker, 2021-12-02:
Hello? Anybody?

frollard, 2021-12-18:
it sounds like your pid is okay for small accelerations but reversing is wrong?