bigtreetech/BIGTREETECH-Stepper-Motor-Driver #19: Klipper setting guide for S42C

salbang, 2022-10-23:
Could you please add setting guide for klipper with corexy kinematics? How to define full_steps_per_ration, microsteps and etc?
Thanks.

kbrouder, 2022-10-26:
Did you have any success? I have 2 waiting to test out, but after reading frustrated users I haven't tried yet....

smooreace, 2023-03-21:
5 months later and still absolutely nothing. I cant find anyone that has documented even trying to set this up. Kinda pisses me off that I just re-wired and loomed my entire printer with the 42C cables only to find that they seem to be totally abandoned

Thanks BTT!

aschafer14, 2023-03-27:
Here are my klipper settings on an octopus pro, you'll have to update you're rotation distance based on your gearing/belt setup. See docs here https://www.klipper3d.org/Rotation_Distance.html . I also had to configure the steppers to "VERY HIGH" on the LED screen and reset.

This is working for the moment but there are some pretty big limitations if you want to maximize print speed. My max_velocity is limited to 350 mm/s and max_accel is at 3000 mm/s^2 anything higher and the motors wont drive the printhead in a straight line. It's like their fighting to correct each other or the processing isn't fast enough. I have done a lot of searching and no one seems to have a solve for this issue. Most have given up and gone back to more traditional drivers.

`[stepper_x]
step_pin: PF13
dir_pin: !PF12
enable_pin: !PF14
rotation_distance: 40
microsteps: 16
full_steps_per_rotation: 200 
endstop_pin: PG6
position_min: -1
position_endstop: 0
position_max: 300 
homing_speed: 60  
homing_retract_dist: 5

[stepper_y]
#MKS Servo42C Motors
step_pin: PG0
dir_pin: PG1
enable_pin: !PF15
rotation_distance: 40 
microsteps: 16 
full_steps_per_rotation: 200  
endstop_pin: PG9
position_min: 0
position_endstop: 300
position_max: 300 
homing_speed: 60  
homing_retract_dist: 5
homing_positive_dir: true`