bigtreetech/BIGTREETECH-Stepper-Motor-Driver #20: S42C DOES NOT WORK CORRECTLY!!!

forcegl, 2023-01-24:
Dear Bigtreetech. Please make a working firmware for S42C and post the sources on Github... At the moment this shit does not want to work at all, it does not pass calibration and does not work with the control board at all, for which I paid money, I don’t know yet. At the calibration stage, an Error Dire error. Everything is connected according to the instructions, the magnet is according to the instructions.

Anuril, 2023-04-15:
Despite your rudeness towards bigtreetech, which I don't like, I can understand your frustration. Please refrain from using that kind of language. 

To solve your Problem, you need to switch the motor connector wire pairs. (Not Cross them!) 
[image]

jcollasius, 2024-09-03:
I had the same error "Error Dire". On top of that the motor stutters on the calibration.
On my sets the cables had no colors and the middle was crossed like this:
1 2 3 4
I   X   I
1 3 2 4
My solution was to change the outer two cables "1" and "4".
The motor now calibrates and runs smoothly.

Anyone googling for this error like I did: Have a nice day, I hope this fixes your error.