bigtreetech/BIGTREETECH-Stepper-Motor-Driver #29: S42C shift Y layer with Marlin 2.1.2.4 and SKR1.3 motherboard

moriamoria, 2024-09-26:
Good moring to all,

I am found of BIGTREE product, and during the process of upgrade a Ortur 4 printer with following components I have a printing issue with my new S42C driver.

on each layer, y(layer+1)=y(layer) -delta

attach a photo of a speed tower test. Looks like Pise with an oriantation of 45°
[image]

I have follow the S42C documentation

bellow my configuration

Hardware : 
 
I have a ortur 3D printer
I have mks 1.3 motherboard
On X, Z1 and Z2 I have 2209 driver
On Y, I have install           BIGTREETECH S42C V1.1
 
Software:
 
I have recompiled Marlin 2.1.2.4
 
I have change only X, Z, Z2 to BIGTREETECH  2209 driver
 
Before I got also a 2209 on Y, but I was missing steps when printing high speed.
 
Installation :
 
I have installed on Y BIGTREETECH S42C V1.1 with adaptor on the motherboard.
 
I have done calibration and looks ok.
 
When I push Y, the stepper get is position again.
 
The Issue : 
 
My print on each layer y(n+1) = y – 0.1mm
 
So, my print looks like babel tower !
 
Can you guide my to solve this issue ?
 
Question : 
 
What is the maximum speed and acceleration for BIGTREETECH S42C V1.1and nema 17 40mm stepper.
 
Regards

moriamoria, 2024-09-26:
In the documentation, it sais 
The firmware of the motherboard only needs to configure the pulse number per millimeter to ensure that it is consistent with the microsteps set in S42C. S42C does not communicate with the motherboard through SPI or UART. Other configurations can be set according to A4988 STANDALONE mode.

So why in the S42c, you can select the baudrate , currently it is 9600, in Marling do I need to configure ?

Can it be the based reason of my bad print ?

I have also print a cube, and same problem.

Regards

ellensp, 2024-10-21:
This sort of lean is often caused by the step pin being inverted.

eg  driver expects _|_  and movea when the signal is high, but it gets ¯|¯   so it steps twice.

moriamoria, 2024-10-22:
Hello, Thanks for your help.

Could you tel me how to configure marlin in order to not have setp pin inverted ?

Regards

ellensp, 2024-10-23:
In Configuration_adv.h

In the latest code

```CPP
// By default stepper drivers require an active-HIGH signal but some high-power drivers require an active-LOW signal to step.
#define STEP_STATE_X HIGH
#define STEP_STATE_Y HIGH
#define STEP_STATE_Z HIGH
#define STEP_STATE_I HIGH
#define STEP_STATE_J HIGH
#define STEP_STATE_K HIGH
#define STEP_STATE_U HIGH
#define STEP_STATE_V HIGH
#define STEP_STATE_W HIGH
#define STEP_STATE_E HIGH
```

In older code

```cpp
// By default pololu step drivers require an active high signal. However, some high power drivers require an active low signal as step.
#define INVERT_X_STEP_PIN false
#define INVERT_Y_STEP_PIN false
#define INVERT_Z_STEP_PIN false
#define INVERT_E_STEP_PIN false
```

moriamoria, 2024-10-23:
Hello,

Many thanks, I will test this before sending back to amazon.

Regards