bigtreetech/BIGTREETECH-Stepper-Motor-Driver #33: TMC2209 V1.3 Sketch layout is differ from the real PCB

timetheonce, 2025-03-04:
Hi Bigtreetech,

The Spread Pin (Pin7) in the sketch is differ from what I got, see the figures below. The real product leave the Spread Pin unconect, while the sketch shows it is pulled high through R7.

Plus, please add instruction of switching between SpreadCycle and SteathChop mode as what have been done in the manual of V1.2.

[image]
[image]

looxonline, 2025-03-04:
NC means "not connected". The diagram and the PCB match.

timetheonce, 2025-03-04:
LOL,, I thought it was Normally Closed.. 
Don't border, thanks

closed 2025-03-04 by timetheonce

looxonline, 2025-03-04:
It can also mean normally closed but only when it is used with a relay. For other uses it usually means not connected.